KIT-ISAS/lili-om #2: data set  download very very slow and

narutojxl, 2020-11-07:
Hi author,
Thanks for your hard work and sharing!  
When download data set, its speed is very very slow and the network always  disconnected from time to time. I don't know why this happened, although I can go online normally.  Could you please share the data set, for example the smallest bag "KA_Urban_Campus_2.bag | 2020-10-29 12:30 | 580M" onto google drive to run the demo? Thanks for your help!
Best

kailaili, 2020-11-07:
Hi, thank you for your interest in the work.
We didn't observe such issues. The download speed is actually quite fast. You may check you r local internet service.
However, I've added a second link on OneDrive there. Hope that could help you. Cheers.

kailaili, 2020-11-07:
I'm going to close this issue..

closed 2020-11-07 by kailaili