KIT-ISAS/lili-om #4: find corresponding features

narutojxl, 2020-11-11:
Hi @kailaili, thanks for your excellent work !
* Q1. When found corresponding features, 
https://github.com/KIT-ISAS/lili-om/blob/97b635225ce55d02ecf28bc66634541530f9fff4/LiLi-OM-ROT/src/BackendFusion.cpp#L1394-L1399
the `for` loop why not is 
```
        for (int i = 0; i < edge_lasts_ds[idx + 1]->points.size(); ++i) {
            pt_in_local = edge_lasts_ds[idx + 1]->points[i];
```
The similar doubt for 
https://github.com/KIT-ISAS/lili-om/blob/97b635225ce55d02ecf28bc66634541530f9fff4/LiLi-OM-ROT/src/BackendFusion.cpp#L1464-L1469

* Q2.  Why not  `lp = q_last_curr * p_w + t_last_curr;`
https://github.com/KIT-ISAS/lili-om/blob/97b635225ce55d02ecf28bc66634541530f9fff4/LiLi-OM-ROT/include/factors/LidarKeyframeFactor.h#L45

Thanks for help and time!
Best

closed 2020-11-11 by kailaili

narutojxl, 2020-11-12:
Hi @kailaili and all,
* I'm very sorry for the Q1 question. I mistake the `vector<int> keyframe_idx;`, its content begins from `1`, not `0`. Sorry again !
* For Q2 question, i give some my reasons about the guess. If there are some mistakes, please tell me, look forward to disscuss with you.
https://github.com/KIT-ISAS/lili-om/blob/97b635225ce55d02ecf28bc66634541530f9fff4/LiLi-OM-ROT/include/factors/LidarKeyframeFactor.h#L28-L54
     * `cp` point is laser's point in current laser's  frame.  `q` and `t` is the corresponding time the imu's pose in the global frame `map`, namely the `imu0` frame.  we use the extrinsic `qlb` and `tlb` between laser and imu, transform `cp` into imu's frame, got `p_w`. then we use `q` and `t` transform `p_w` into  `map` frame.  `lpa` and `lpb` are the corresponding edge features in `map`.
    * I notice that  calculate surf points's residual, we use the `point_w`, not the `cp` point.
 https://github.com/KIT-ISAS/lili-om/blob/97b635225ce55d02ecf28bc66634541530f9fff4/LiLi-OM-ROT/include/factors/LidarKeyframeFactor.h#L89-L104

limeng1523, 2020-11-12:
Hi @narutojxl, thanks for your interest in this work. I'm sorry to reply to you so late. I checked the code, and you are right about Q2 question. This is a typo. It should be p_w  instead of cp. These large residuals are suppressed by the Cauchy kernel function, so I didn't find this typo. I'm really sorry to make this typo, and thanks for pointing it out. I have fixed it, and we will update the code soon. If you have other questions, welcome to discuss with us.

narutojxl, 2020-11-13:
@limeng1523,  thanks very much ! My problen is solved.