KIT-ISAS/lili-om #5: which metis version are you using

YushengWHU, 2020-11-11:
Hello here, when correctly complied my lili_om, I wanna launch with example bag, but the terminal gives me
```
devel/lib/lili_om/BackendFusion: error while loading shared libraries: libmetis.so: cannot open shared object file: No such file or directory
[ INFO] [1605101464.753388799]: ----> Preprocessing Started.
[ INFO] [1605101464.769086095]: ----> Lidar Odometry Started.
[BackendFusion-7] process has died [pid 19832, exit code 127, cmd /home/yusheng/Documents/3rd_party/livox/ws_livox/devel/lib/lili_om/BackendFusion __name:=BackendFusion __log:=/home/yusheng/.ros/log/2564d1c6-2422-11eb-9512-94b86da529ef/BackendFusion-7.log].
log file: /home/yusheng/.ros/log/2564d1c6-2422-11eb-9512-94b86da529ef/BackendFusion-7*.log
[ WARN] [1605101494.615353143]: Waiting for IMU data ...
```
Which shows I have not yet installed a metis, thus I wanna know which version of metis are you using?

limeng1523, 2020-11-11:
@YushengWHU 
Hi, thank you for your interest in our work. Actually I'm not true the version of libmetis.so we used. This error may have been caused by GTSAM. I will give you some methods to tackle it.

1. You need to check if there is a file named 'libmetis.so' in '/usr/local/lib'. If not, try to install it using 
    sudo apt-get update -y
    sudo apt-get install -y libmetis-dev

2. If method 1 dosen't work, you can check the path '/usr/lib'. Sometimes libmetis.so is installed in '/usr/lib'. Just copy    libmetis.so to '/usr/local/lib'.

3. If method 1 and 2 don't work. You can try to reinstall GTSAM with another version.

I hope this can help you. If you still can't solve error, maybe I can send the file 'libmetis.so' to you by email. Just put it to '/usr/local/lib', the error should be solved.

YushengWHU, 2020-11-12:
Thanks for the quick response, I have solved the problem with your method 1.
However, the BackendFusion process still died whenever I launch the system, and no matter which launch file am using, same problem still remains. With
```
setting /run_id to 4644a900-2483-11eb-9455-94b86da529ef
process[rosout-1]: started with pid [3281]
started core service [/rosout]
process[rviz-2]: started with pid [3288]
process[lili_om-3]: started with pid [3289]
process[FormatConvert-4]: started with pid [3290]
process[Preprocessing-5]: started with pid [3296]
[ INFO] [1605143181.084842236]: ----> Format Convert Started.
process[LidarOdometry-6]: started with pid [3302]
process[BackendFusion-7]: started with pid [3303]
[ INFO] [1605143181.143823662]: ----> Preprocessing Started.
[ INFO] [1605143181.149683014]: ----> Lidar Odometry Started.
[BackendFusion-7] process has died [pid 3303, exit code -11, cmd /home/yusheng/Documents/3rd_party/livox/ws_livox/devel/lib/lili_om/BackendFusion __name:=BackendFusion __log:=/home/yusheng/.ros/log/4644a900-2483-11eb-9455-94b86da529ef/BackendFusion-7.log].
log file: /home/yusheng/.ros/log/4644a900-2483-11eb-9455-94b86da529ef/BackendFusion-7*.log
```
I checked my logfile, and the error is
```
[rosmaster.threadpool][ERROR] 2020-11-12 09:07:28,685: Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rosmaster/threadpool.py", line 218, in run
    result = cmd(*args)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rosmaster/master_api.py", line 210, in publisher_update_task
    ret = xmlrpcapi(api).publisherUpdate('/master', topic, pub_uris)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1243, in __call__
    return self.__send(self.__name, args)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1602, in __request
    verbose=self.__verbose
  File "/usr/lib/python2.7/xmlrpclib.py", line 1283, in request
    return self.single_request(host, handler, request_body, verbose)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1316, in single_request
    return self.parse_response(response)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1493, in parse_response
    return u.close()
  File "/usr/lib/python2.7/xmlrpclib.py", line 800, in close
    raise Fault(**self._stack[0])
Fault: <Fault -1: 'publisherUpdate: unknown method name'>
```
Some suggestions including python version conflicts, as I just updated to Ubuntu 18.04 recently and only had system python alone, I thought this might not be the solution. One similar open issue is [issue](https://github.com/ros/ros_comm/issues/572).

So do you have any suggestions?

YushengWHU, 2020-11-12:
By the way, the setup of the system strictly follows your instructions, with Ceres 2.0, GTSAM 4.0.0

kailaili, 2020-11-12:
I've had such problem before due to system/ROS update and decided not to give much thought to it. The solution is just to reinstall Ubuntu and ROS totally. It's not relevant to the instructions that we gave.

YushengWHU, 2020-11-12:
Everything solved with reinstallation.

closed 2020-11-12 by YushengWHU