KIT-ISAS/lili-om #7: build local  graph optimization

narutojxl, 2020-11-13:
Hi all, 
First of all, I would like to express my high respect and thanks to the authors's  hard work!
I will try to carefully describe my problem clearly, please be patient.
* I don't set modify the slide window size, so the default value is still 3, there are 3 keyframes in the current window.
* If we set the first received laser's index is 1, by the backend, then we will perform the first slide window when we received 3rd lasers, perform the second slide window when received 4th lasers. 
* When we use the imus between the first keyframe in the current window and the last keyframe to integrate to get `pose_each_frame[]` and `pose_info_each_frame[]`, then we use this two variables as the initial values to build the local optimization graph: 
I found that the start integration keyframe `Ps[], Vs[], Rs[]` shoule be  `Ps[Ps.size() - slide_window_width - 1]`, the `Ps[Ps.size() - slide_window_width]` is the first keyframe in the current window. I think it should be a typo, agree ?
https://github.com/KIT-ISAS/lili-om/blob/4ab3639cbfd58a7f3aa1bde5f8277d2c64579076/LiLi-OM-ROT/src/BackendFusion.cpp#L1654-L1661
* I print the `Ps[]` value as following, thanks for your help and time!
>
	[ INFO] [1605269721.485954907]: ----> Preprocessing Started.
	[ INFO] [1605269721.502283278]: ----> Back End Started.
	[ INFO] [1605269721.528559733]: ----> Lidar Odometry Started.
	laser_index = 1 start
	laser_index = 1 done!

	laser_index = 2 start
	laser_index = 2 done!

	laser_index = 3 start
	laser_index = 3 done!

	laser_index = 4 start
	curr Ps[].size = 5
	Ps[0] = 0.0000, 0.0000, 0.0000
	Ps[1] = 0.0125, -0.0138, 0.0002
	Ps[2] = 0.0174, -0.0197, 0.0008
	Ps[3] = 0.0165, -0.0414, -0.0004
	Ps[4] = 0.0241, -0.0444, 0.0008
	laser_index = 4 done!

	laser_index = 5 start
	curr Ps[].size = 6
	Ps[0] = 0.0000, 0.0000, 0.0000
	Ps[1] = 0.0125, -0.0138, 0.0002
	Ps[2] = 0.0174, -0.0197, 0.0008
	Ps[3] = 0.0152, -0.0412, 0.0003
	Ps[4] = 0.0211, -0.0393, 0.0010
	Ps[5] = 0.0251, -0.0362, -0.0002
	laser_index = 5 done!

kailaili, 2020-11-14:
it's not.

closed 2020-11-14 by kailaili

narutojxl, 2020-11-16:
Hi all, 
I add some information about the guess. I don't figure out where i mistake yet :( , look for help for anybody. I will be much appreciated.

>
        if (pose_cloud_frame->points.size() == slide_window_width) {
            pose_each_frame->push_back(pose_cloud_frame->points[0]); 
            pose_info_each_frame->push_back(pose_info_cloud_frame->points[0]);
        } else if(pose_cloud_frame->points.size() > slide_window_width) {
            int ii = imu_idx_in_kf[imu_idx_in_kf.size() - slide_window_width - 1]; 
            double dx = 0, dy = 0, dz = 0, rx = 0, ry = 0, rz = 0;

            Eigen::Vector3d Ptmp = Ps[Ps.size() - slide_window_width]; 
            Eigen::Vector3d Vtmp = Vs[Ps.size() - slide_window_width];
            Eigen::Matrix3d Rtmp = Rs[Ps.size() - slide_window_width]; 
            Eigen::Vector3d Batmp = Eigen::Vector3d::Zero();
            Eigen::Vector3d Bgtmp = Eigen::Vector3d::Zero();
  
            std::printf("pose_cloud_frame.size= %d, ii= %d, tmp= %d\n", 
                                  pose_cloud_frame->points.size(), 
                                  imu_idx_in_kf.size() - slide_window_width - 1, 
                                  Ps.size() - slide_window_width); 

          //pose_cloud_frame.size= 4,  ii= 0,  tmp= 2
 
[image]
`ii` is imu's index, in `imu_buf[ii]`, imu's measuremens  are green lines. `i` is laser's index in `each_odom_buf[i]`, its start and end position as the picture shows. `Rtmp` and `Ptmp` are the `Ps[2]`and `Ptmp[2]`, because ` Ps.size() - slide_window_width = 2`, when  after performed the second slide window and ready to build local graph map. 
It seems that timestamp of `Rtmp` is older than timestamp of `imu_buf[ii]`, but in the `while` loop, the `Rtmp` is the start integration moment. I am not sure whether i totally understand the code correctly, not mistake and not miss anything, it confused me.  Thanks for your time again !
https://github.com/KIT-ISAS/lili-om/blob/4ab3639cbfd58a7f3aa1bde5f8277d2c64579076/LiLi-OM-ROT/src/BackendFusion.cpp#L1682-L1717