KIT-ISAS/lili-om #9: backend error

narutojxl, 2020-11-16:
Hi all
Sometimes when i run  `rosbag play FR_IOSB_Tree_64.bag  --clock  -r 1 --pause`, everything is OK. The terminal output nothing, and the map showed in rviz is also OK.  Sometimes  the rviz result is unnormal and error occured about the backend, it seems about rotation?  Did anyone else encounter this problem?
>
	jxl@dell:~$ roslaunch lili_om_rot run_fr_iosb.launch 
	... logging to /home/jxl/.ros/log/5b5130ec-2803-11eb-a3af-8cec4bc262f3/roslaunch-dell-1060.log
	Checking log directory for disk usage. This may take awhile.
	Press Ctrl-C to interrupt
	Done checking log file disk usage. Usage is <1GB.

	started roslaunch server http://localhost:36597/

	SUMMARY
	========

	PARAMETERS
	 * /IMU/acc_n: 2000
	 * /IMU/acc_w: 2
	 * /IMU/gyr_n: 0.0173
	 * /IMU/gyr_w: 0.00025
	 * /backend_fusion/imu_topic: /applanix/imu
	 * /backend_fusion/kd_max_radius: 1.0
	 * /backend_fusion/lc_icp_thres: 0.2
	 * /backend_fusion/lc_map_width: 25
	 * /backend_fusion/lc_search_radius: 15
	 * /backend_fusion/lc_time_thres: 60.0
	 * /backend_fusion/lidar_const: 7.5
	 * /backend_fusion/local_map_width: 50
	 * /backend_fusion/loop_closure_on: True
	 * /backend_fusion/mapping_interval: 2
	 * /backend_fusion/max_num_iter: 15
	 * /backend_fusion/ql2b_w: 0.7071
	 * /backend_fusion/ql2b_x: 0
	 * /backend_fusion/ql2b_y: 0
	 * /backend_fusion/ql2b_z: 0.7071
	 * /backend_fusion/save_pcd: True
	 * /backend_fusion/slide_window_width: 3
	 * /backend_fusion/surf_dist_thres: 0.12
	 * /backend_fusion/tl2b_x: -0.18
	 * /backend_fusion/tl2b_y: 0
	 * /backend_fusion/tl2b_z: -0.095
	 * /common/data_set: iosb
	 * /common/frame_id: lili_om_rot
	 * /lidar_odometry/if_to_deskew: False
	 * /lidar_odometry/max_num_iter: 12
	 * /lidar_odometry/scan_match_cnt: 1
	 * /preprocessing/ds_rate: 4
	 * /preprocessing/lidar_topic: /velodyne_pcl_gen...
	 * /preprocessing/line_num: 64
	 * /rosdistro: kinetic
	 * /rosversion: 1.12.14
	 * /use_sim_time: False

	NODES
	  /
	    BackendFusion (lili_om_rot/BackendFusion)
	    LidarOdometry (lili_om_rot/LidarOdometry)
	    Preprocessing (lili_om_rot/Preprocessing)
	    lili_om_rot (tf/static_transform_publisher)
	    rviz (rviz/rviz)

	ROS_MASTER_URI=http://localhost:11311

	process[rviz-1]: started with pid [1081]
	process[lili_om_rot-2]: started with pid [1082]
	process[Preprocessing-3]: started with pid [1083]
	process[LidarOdometry-4]: started with pid [1095]
	process[BackendFusion-5]: started with pid [1096]
	[ INFO] [1605531281.625058651]: ----> Back End Started.
	[ INFO] [1605531281.631377798]: ----> Lidar Odometry Started.
	[ INFO] [1605531281.641533446]: ----> Preprocessing Started.
	F1116 20:54:55.148710  1096 problem_impl.cc:82] Check failed: !RegionsAlias(existing_block, existing_block_size, new_block, new_block_size) Aliasing detected between existing parameter block at memory location 0x7fff5dc4a950 and has size 3 with new parameter block that has memory address 0x7fff5dc4a960 and would have size 4.
	*** Check failure stack trace: ***
	    @     0x7f52d8ee243d  google::LogMessage::Fail()
	    @     0x7f52d8ee4253  google::LogMessage::SendToLog()
	    @     0x7f52d8ee1fcb  google::LogMessage::Flush()
	    @     0x7f52d8ee4c3e  google::LogMessageFatal::~LogMessageFatal()
	    @           0x530f0c  ceres::internal::(anonymous namespace)::CheckForNoAliasing()
	    @           0x533d8e  ceres::internal::ProblemImpl::InternalAddParameterBlock()
	    @           0x53438c  ceres::internal::ProblemImpl::AddResidualBlock()
	    @           0x534e0c  ceres::internal::ProblemImpl::AddResidualBlock()
	    @           0x4b1ee0  BackendFusion::optimizeLocalGraph()
	    @           0x4f1b4b  BackendFusion::saveKeyFramesAndFactors()
	    @           0x495186  main
	    @     0x7f52d6eb2830  __libc_start_main
	    @           0x495d59  _start
	[BackendFusion-5] process has died [pid 1096, exit code -6, cmd /home/jxl/lili_om_ws/devel/lib/lili_om_rot/BackendFusion __name:=BackendFusion __log:=/home/jxl/.ros/log/5b5130ec-2803-11eb-a3af-8cec4bc262f3/BackendFusion-5.log].
	log file: /home/jxl/.ros/log/5b5130ec-2803-11eb-a3af-8cec4bc262f3/BackendFusion-5*.log
	^C[LidarOdometry-4] killing on exit
	[Preprocessing-3] killing on exit
	[lili_om_rot-2] killing on exit
	[rviz-1] killing on exit
	shutting down processing monitor...
	... shutting down processing monitor complete
	done
	jxl@dell:~$ 

This is the rviz's snap capture.
[image]

I modify the CMakeLists `set(CMAKE_BUILD_TYPE "Release")`, my OS is ubuntu16.04, installed eigen3.3.7, ceres2.0.0 and gtsam-4.0.2, config as following. I don't modify the config file and launch file. Although play bag, we also published `--clock`, but i see the launch file `<param name="/use_sim_time" value="false" />`, i remain the param `false`.
```
# find_package(Eigen3 REQUIRED) #default
list(APPEND CMAKE_PREFIX_PATH "/home/jxl/third_softwares/eigen_3_3_7/INSTALL_DIR/share/eigen3/cmake")
#set(CMAKE_PREFIX_PATH "/home/jxl/third_softwares/eigen_3_3_7/INSTALL_DIR/share/eigen3/cmake")
find_package(Eigen3 3.3.7 REQUIRED)

# find_package(Ceres REQUIRED) #default
find_package(Ceres REQUIRED PATHS "/home/jxl/third_softwares/ceres-solver-2.0.0/INSTALL/")

find_package(GTSAM REQUIRED)
```

kailaili, 2020-11-16:
We didn't test it under Ubuntu 16.04 as written in the instruction. Also, we didn't observe such an issue.

closed 2020-11-16 by kailaili

LeisureLei, 2021-06-02:
> We didn't test it under Ubuntu 16.04 as written in the instruction. Also, we didn't observe such an issue.

How do you fix this issue? I meet same problem with you!

xingyuuchen, 2022-12-29:
The same rviz result, but I didn't see the crash.