KIT-ISAS/lili-om #12: Inconsistent performance from different start timestamp using the same dataset

tanmengwen, 2020-12-17:
Hi kailaili,
Thanks for your good work!

I compiled and run this project with the given datatset, for example FR_IOSB_Tree_64.bag.
It was working well when i played the rosbag from the begin time. But it drifted quite a lot when i played the rosbag from a later timestamp such as from 30s.

I didn't know what reason caused this result. I guess maybe it should start with static status. Looking for forward your reply.
Thank you!

[image]

kailaili, 2020-12-17:
[image]

Hi, above is the result from running lili_om_rot on this sequence. There shouldn't be a problem. Maybe you should check whether you used the right launch file (run_fr_iosb.launch).

closed 2020-12-17 by kailaili

tanmengwen, 2020-12-17:
Hi kailaili,

I'm sure I used the right launch file(run_fr_iosb.launch). 
I could also get abover result same as you when play the rosbag from the start timestamp.  

I mean lili_om_rot would drift, for example, just use the command below to jump the head 30s. 
rosbag play FR_IOSB_Tree_64.bag -r 1.0 --clock  -s 30

kailaili, 2020-12-17:
Apparently you need to have a proper initialization of the IMU. The current one initializes the IMU given a static state.