KIT-ISAS/lili-om #13: Evaluation

Liming-Cheng, 2020-12-20:
Hi @kailaili , it's nice to see lili-om paper shows evaluation with LINS on UrbanNav dataset, and I'm confused how to set the imu parameters in the LINS config files because its unit seems different with normal (m/(s^2), rad/s).  Do you have any suggestions? Thanks a lot!

kailaili, 2020-12-20:
I don't really remember that we have special treatment for that config file. If you are unsure about that, why not ask the author of LINS.

closed 2020-12-20 by kailaili

Liming-Cheng, 2020-12-26:
Dear @kailaili ,
Sorry to bother you again. I have asked the author of LINS, but not got a solution. 
Here is the issue. [https://github.com/ChaoqinRobotics/LINS---LiDAR-inertial-SLAM/issues/27](url)
Fortunately，your paper shows LINS's result on Urbannav dataset, and it seems only you can help with that. 
Can you have a review of your testing on Urbannav when you are free? That's verry important, thank you!

kailaili, 2020-12-26:
As I said, we didn't have special treatment to the config file of LINS. I can't help you further in this regard.

Liming-Cheng, 2020-12-26:
Dear@kailaili , Thanks for your quick reply!  I set the parameters according to Urbannav dataset as the file shows, and I do not change them, but it runs with big drift and even error.
Here is my config file picture. 
[image]
I don't know whether there is some incorrect operations. Can you show me your config file? Thank you!

kailaili, 2020-12-26:
The answer given by the author of LINS completely makes sense. You may even tune these parameters by yourself. Otherwise, there's nothing to show you from our side. Sorry.