KIT-ISAS/lili-om #15: LiLi-OM-ROT BackendFusion node died where isam->update()

wwtinwhu, 2020-12-26:
First,  Thanks your excellent work and well-written paper!  
I test the LiLi-OM-ROT using FR_IOSB_Tree_64.bag with loop closure on. I found that the BackendFusion died in saveKeyFramesAndFactors() where isam->update().  The more amazing thing is that in the terminal, there is no error output. I use cout to find that problem. 
So,  Is this the reason that gtsam's version? I use gtsam-4.0.0-alpha2.

kailaili, 2020-12-26:
We don't observe this issue. And we don't use the version of gtsam you mentioned here. That sequence runs well if dependencies are correctly installed. 

[image]

closed 2020-12-26 by kailaili

wwtinwhu, 2020-12-26:
> We don't observe this issue. And we don't use the version of gtsam you mentioned here. That sequence runs well like as below if dependencies are correctly installed (you may try the most recent stable release of gtsam like 4.0.2 or 4.0.3).
> 
> [image]

Could you share the specific version of Eigen and GTSAM you used? Thank you very much!

kailaili, 2020-12-26:
Installation of Eigen comes with Ceres as one of its dependencies. Otherwise, all required dependencies of lili-om are clearly stated in readme.

wwtinwhu, 2020-12-28:
> Installation of Eigen comes with Ceres as one of its dependencies. Otherwise, all required dependencies of lili-om are clearly stated in readme.

Thank you . For now, I can not run the LiLi-OM-ROT successfully in loop-closure on . I will try different eigen version and gtsam version until it works. And Update this issue.

wwtinwhu, 2020-12-29:
> > Installation of Eigen comes with Ceres as one of its dependencies. Otherwise, all required dependencies of lili-om are clearly stated in readme.
> 
> Thank you . For now, I can not run the LiLi-OM-ROT successfully with loop-closure on . I will try different eigen version and gtsam version until it works. And Update this issue.

Ubuntu18.04, Eigen 3.3.7, Ceres 2.0.0, gtsam 4.0.2. It works.  
[image]