KIT-ISAS/lili-om #16: LiLi-OM-ROT BackendFusion node died randomly using own data

wwtinwhu, 2020-12-29:
Hello! I test LiLi-OM-ROT with my own data. Sometimes, the BackendFusion node will die in the function called saveKeyFramesAndFactors(). The specific code line where it dies is as follows.
`if(imu_buf[i]->header.stamp.toSec() > timeodom_cur)`
By the way, this error happens randomly. Do you have any suggestions? Thank you very much!

kailaili, 2020-12-29:
I suppose you just need to tune the parameters according to your own scenario as well as hardware. Also pay attention to your time synchronization. The BackendFusion node cannot die if things are set up properly.

closed 2020-12-29 by kailaili

wwtinwhu, 2020-12-29:
> I suppose you just need to tune the parameters according to your own scenario as well as hardware. Also pay attention to your time synchronization. The BackendFusion node cannot die if things are set up properly.

Thanks for your advice! I will try to figure it out.

wwtinwhu, 2021-01-03:
> > I suppose you just need to tune the parameters according to your own scenario as well as hardware. Also pay attention to your time synchronization. The BackendFusion node cannot die if things are set up properly.
> 
> Thanks for your advice! I will try to figure it out.

This may be caused by the code in the function imuHandler() where the specific line is as follows.
`imu_buf[imu_buf.size() - 601] = nullptr;`
I comment this line and this error hasn't happened yet. I think it is relevant to the type of this shared_ptr which is the const type. Need professional answers！

pangchenglin, 2021-02-03:
> > > 我想您只需要根据自己的情况以及硬件来调整参数。也要注意您的时间同步。如果一切设置正确，BackendFusion节点将不会消失。
> > 
> > 
> > 谢谢你的建议！我会设法弄清楚。
> 
> 这可能是由于imuHandler（）函数中的代码引起的，其中的特定行如下。
> `imu_buf[imu_buf.size() - 601] = nullptr;`
> 我对此行发表评论，但尚未发生此错误。我认为这与此shared_ptr的类型（即const类型）有关。需要专业答案！

I meet the same error.Did you fix this problem?

pangchenglin, 2021-02-03:
> This may be caused by the code in the function imuHandler() where the specific line is as follows.
> `imu_buf[imu_buf.size() - 601] = nullptr;`
> I comment this line and this error hasn't happened yet. [...]

I meet the same error.Did you fix this problem?

wwtinwhu, 2021-02-04:
> I meet the same error.Did you fix this problem?

No. For now, I just comment it. Do you have any advice? @pangchenglin

JanQuenzel, 2022-02-20:
The problem stems from trying to access a nullptr. In ```saveKeyFramesAndFactors``` it accesses ```imu_buf``` without checking if there still is a valid measurement at ```imu_idx```. Only upto the last 600 imu msg ptrs in imu_buf contain valid pointers. A higher imu data rate would fill in faster and result in imu_buf[imu_idx] being a nullptr.
Commenting the previously mentioned line prevents setting older ones to nullptr, hence no apparent problem.
BUT the std::vector of ```ìmu_buf``` (as well as many others like biases in backendfusion) never shrinks, so running it on huge datasets is memory limited. For this kind of max-size buffers, it would be better to use a std::deque instead. Changing it though won't be as easy, since creating indices from ```someVector.size()-someOtherSize``` is quite often used throughout the backendfusion.