KIT-ISAS/lili-om #17: Drift Problem with Marginalization Factor

kamibukuro5656, 2021-01-21:
Hello.

Thank you for your help with the topic on built-in IMUs.

I have been using the built-in IMU and LILI-OM to map the data that I have collected.
However, there were some situations where the data drifted rapidly and could not be maped properly.
The following picture shows a SLAM of the 2020-openload data published by Livox using the built-in IMU and LILI-OM, but the data drifted in the opposite direction in the middle.
[image]

I thought that the cause of this phenomenon was one of the residual terms, so I commented out the registrations to the ceres solver one by one to isolate the problem.
As a result, commenting out the MarginalizationFactor resulted in the following good results. (I changed "true" to "false" in line 833 of BackendFusion.cpp)

[image]

I would like to know if you have any advice about this result.

kailaili, 2021-01-21:
Hi, I would suggest some further tuning of the parameter. Disabling the marginalization may give you a converged result, but it also discard the motion information between sliding windows. The problem perhaps does not really come from marginalization, but rather incorrect optimization results in the backend fusion. And this might result from an insufficient quality of inertial data or feature points.

kamibukuro5656, 2021-01-22:
Thank you for your answer.

I understand now.

The difference with your published data is that you use the built-in IMU, so I assume that the problem is more with the quality of the inertial data than the feature points.

Are there any parameters that should be adjusted when using a low accuracy IMU like the built-in IMU?
Also, is it conceivable to add new weight coefficients to each residual term to adjust the effect?

kailaili, 2021-01-22:
Hi, thank you for sharing your experience. 
Yes, we used a separate IMU for inertial measurements. I assume, in your case, you could try giving a large weights to the Lidar residual? Or changing the downsampling ratio for the feature map? These parameters can be changed in the yaml file.

closed 2021-01-24 by kailaili