KIT-ISAS/lili-om #25: There are some problems in the initial trajectory of rotating lidar data

Soverngity, 2021-02-17:
Hello author, thank you for your work.
When I was running my own pandar40 data, there was such a problem with the initial trajectory point in rviz.
[image]

kailaili, 2021-02-17:
We didn't test lili-om on Pandar40. There could be many reasons for such an effect. You may check you own definition of coordinates, your calibrations, and tune the parameters accordingly.

closed 2021-02-17 by kailaili

Soverngity, 2021-03-18:
Dear author, I solved the above problem because of the acceleration of gravity.
But now I have another problem, as shown in the figure.The code runs normally in the previous period of time, but every time it runs for a long time, the system will report an error. It seems to be the problem of Ceres error term. Do you have any opinion about it?
[image]

Soverngity, 2021-03-18:
@kailaili