KIT-ISAS/lili-om #27: Use Livox Avia

duanqg, 2021-04-29:
Good job！
Can I use Livox Avia to run the code？

kailaili, 2021-04-29:
For the repetitive line scanning of Livox Avia, the system should be applicable given proper tuning of parameters. (We didn't test it yet).

closed 2021-04-29 by kailaili

duanqg, 2021-05-01:
@kailaili 
thank you
Is the avia Non-repetitive Circular Scanning pass your test？
Do the parameters need to be changed？
thank you！

duanqg, 2021-05-01:
Dear Brother Li，I would like to use the non-repeated scanning of Livox Avia
to carry out SLAM of lili. Can I achieve the same effect as Livox Horizon?
If debugging changes are made, where should I make the changes?

kailaili, 2021-05-01:
The system was not tailored for the scan pattern of circular shape. One could replace the feature extraction module with the one from loam_livox or something similar.

duanqg, 2021-05-11:
感谢您！我使用livox avia有了很好的结果，应该是可以适配的。
但是由于MTI680G的原因，出现了拐弯部分飞起偏移的情况，这个在horizon和avia上都出现了。
然后看到了这个issue，https://github.com/KIT-ISAS/lili-om/issues/14，和我的问题一模一样，请问按照这篇文章所说的imu的angular velocity and linear acceleration x and y 在代码的哪个部分改动?

bugdeveloper0, 2022-01-20:
> 

@duanqg 哥们，你现在可以跑适配avia的包了吗

aiqibaobei, 2022-12-28:
@bugdeveloper0  我尝试了自己录的avia包，可以跑的，但是因为作者的代码需要两个imu，用那个internal launch应该就可以了