KIT-ISAS/lili-om #28: How to use it with my own data?

Mikor-mkr, 2021-05-03:
Sorry for the newbie question but I can't get my head around on how to use it with my own data, could somebody clarify this ? How can I subscribe lili_om to my data?

kailaili, 2021-05-04:
Hello, without further details about your own setup we probably are not able to provide meaningful help. In general, you would need to apply you own calibration, change some of the topic names and tune the parameters of the system, etc.

closed 2021-05-04 by kailaili

Mikor-mkr, 2021-05-04:
Hello @kailaili , thanks for your reply. In general I capture 3 topics, a pointcloud2 (vlp16), an imu & 6 dof pose (both coming from T265), I calibrate the 6dof pose sensor and the 3d pointcloud using [lidar_align](https://github.com/ethz-asl/lidar_align) package so I can get a static tf, my question is how to I make lili_om_rot's nodes to subscribe to the suitable topics so the mapping can begin. My first thought was to subscribe through the launch file but when I checked it, I noticed you mention a config file and my guess is that this is how you declare the topics. But I have some questions about the parameters of the yaml file, I think I should create a new thread right ?