KIT-ISAS/lili-om #29: backend error on 18.04, is there need for tuning..?

Mikor-mkr, 2021-05-06:
Hello, I can't get it to run with my own data, I get this error:

```
F0506 14:06:24.649806 32614 problem_impl.cc:84] Check failed: !RegionsAlias( existing_block, existing_block_size, new_block, new_block_size) Aliasing detected between existing parameter block at memory location 0x7ffdc1903b90 and has size 3 with new parameter block that has memory address 0x7ffdc1903ba0 and would have size 4.
*** Check failure stack trace: ***
    @     0x7f4848a7d0cd  google::LogMessage::Fail()
    @     0x7f4848a7ef33  google::LogMessage::SendToLog()
    @     0x7f4848a7cc28  google::LogMessage::Flush()
    @     0x7f4848a7f999  google::LogMessageFatal::~LogMessageFatal()
    @     0x5569ba59b77c  (unknown)
    @     0x5569ba59deff  (unknown)
    @     0x5569ba59f645  (unknown)
    @     0x5569ba50fe67  (unknown)
    @     0x5569ba54b42c  (unknown)
    @     0x5569ba4f9013  (unknown)
    @     0x7f4846d80bf7  __libc_start_main
    @     0x5569ba4f999a  (unknown)
[BackendFusion-5] process has died [pid 32614, exit code -6, cmd /home/antonis/lili_om_ws/devel/lib/lili_om_rot/BackendFusion __name:=BackendFusion __log:=/home/antonis/.ros/log/c362c978-ae51-11eb-805d-e86a640b1f6f/BackendFusion-5.log].
log file: /home/antonis/.ros/log/c362c978-ae51-11eb-805d-e86a640b1f6f/BackendFusion-5*.log
^C[LidarOdometry-4] killing on exit
[Preprocessing-3] killing on exit
[lili_om_rot-2] killing on exit
[rviz-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

Similar problem that #9  mentioned, but I am using 18.04 linux, I think I need some further tuning but I do not know what there parameters are, can you help me with that ?

kailaili, 2021-05-07:
We didn't test our system on VLP-16 yet. It's quite likely that the system cannot work if you directly deploy your own sensor setup without essential modification of the code.

closed 2021-05-07 by kailaili

Mikor-mkr, 2021-06-02:
@JasperTan97 Hello, no I didn't do something, I'm sorry.

JasperTan97, 2021-06-02:
@Mikor-mkr Actually I solved my issue. Turns out the IMU frame and the Lidar frame needed to be aligned. I saw something like this because the IMU was moving forward normal to the ground.

xingyuuchen, 2022-12-29:
> @Mikor-mkr Actually I solved my issue. Turns out the IMU frame and the Lidar frame needed to be aligned. I saw something like this because the IMU was moving forward normal to the ground.

Hello,
Can you be more specific on the solution?
Thanks in advance!