KIT-ISAS/lili-om #33: Backend fusion dying

JasperTan97, 2021-06-02:
I have tried the given datasets for the velodyne version of lili-om and it works.

When I run my own dataset, this occurs. Does anyone know what this means? For reference, the dataset I use is the park dataset from [lio-sam](https://github.com/TixiaoShan/LIO-SAM).

> BackendFusion: /usr/include/boost/smart_ptr/shared_ptr.hpp:734: typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = const sensor_msgs::Imu_<std::allocator<void> >; typename boost::detail::sp_member_access<T>::type = const sensor_msgs::Imu_<std::allocator<void> >*]: Assertion `px != 0' failed.

Thanks everyone!

closed 2021-06-02 by JasperTan97

Bazinga92445, 2022-11-23:
Hi, have you solved this problem eventually? It also occurred when I ran lili_om on  KA_Urban_East.

JasperTan97, 2022-12-06:
No. I am sorry.

xingyuuchen, 2022-12-29:
Hi, I also encounter the same question.

xingyuuchen, 2022-12-29:
Hi, you can see more details of the crash using gdb core dump.