KIT-ISAS/lili-om #35: Running package with Ouster lidar

Jaetriel, 2021-06-18:
Hey, I'm really interested to try out your package and play around with it but I only have access to an OS1-128, is it possible to make this package work with that?

kailaili, 2021-06-18:
Yes. With proper adaptation and tuning, lili-om-rot should be able to work with it.

closed 2021-06-19 by kailaili

vniclas, 2021-07-02:
Hi @Jaetriel, did you manage to adapt the package to be used with the OS1-128 and its internal IMU?

Jaetriel, 2021-07-06:
@vniclas hi, sorry for the late reply. I haven't yet found the time to try and make the package work with Ouster lidars, sorry

vniclas, 2021-07-06:
Thanks for your reply @Jaetriel. Don't worry about it. I made the code run with an Ouster LiDAR by now but haven't attempted any parameter tuning yet.

Jaetriel, 2021-07-06:
@vniclas great work, was it difficult to get it to work?

vniclas, 2021-07-06:
@Jaetriel No, only minimal changes in the code are required. 1) I added the [Ouster PCL type](https://github.com/ouster-lidar/ouster_example/blob/master/ouster_ros/include/ouster_ros/point.h) to `common.h` and 2) the case `N_SCANS == 128` in `Preprocessing.cpp` reading the ring ID directly from the new point type.
Additionally, you'll have to adapt the input topics in the configuration file to use the Ouster point cloud and IMU data. Until now, I only changed the extrinsics but haven't played around with the other parameters yet.
Sometimes, issues related to the _leaf size_ are popping up. Those might be related to some (hard-coded) parameters.