KIT-ISAS/lili-om #37: [Question] How did you time synchronise the IMU & Lidar?

seajayshore, 2021-06-30:
Sorry if this is written somewhere and I just didn’t see.

How are you achieving good synchronisation between the IMU data & Lidar?
A PPS output from the IMU? Do you use GPS clock source?

kailaili, 2021-06-30:
Hi, we used PTP for the synchronization. One possible solution can be found here: https://github.com/ptpd/ptpd

seajayshore, 2021-06-30:
Ah, but PTP makes sense for the Lidar & system.
But the MTI-670 IMU has no PTP capability right? How did you synchronise that?

kailaili, 2021-06-30:
The IMU data is synchronized using the GPS channel onboard

seajayshore, 2021-06-30:
Great, thanks for confirming! :-)

closed 2021-06-30 by seajayshore

asteppke, 2022-01-05:
A quick follow-up on the same topic: the Livox Lidar has different synchronization options, here PTP can be used to synchronize Lidar <-> PC clock. The MTI-670 IMU uses a GPS receiver on the development board, which allows to synchronize IMU <-> GPS. 

How is the remaining synchronization between PC clock <-> IMU/GPS achieved? 
Thank you for any insights.