KIT-ISAS/lili-om #38: about the imu parameters

CanCanZeng, 2021-07-02:
Hi, thank you for sharing your great work!
I tested the given dataset, it runs very well. Then I also run my own data, it's a velodyne HDL 32 lidar with an high precision imu. At the beginning I forgot to set the imu parameters and extrinsics, but it still runs very well. Then I try to set the intrinsics and extrinsics, but I find that you set the acc_n to be a very big value, much bigger than I know, why is it? Does it mean that you make the weight of imu acceleration very small, just use the rotation to aid the lidar slam?

kailaili, 2021-07-03:
Hi, thank you for your information. You may try tuning the parameters given your own hardware setup.

closed 2021-07-03 by kailaili

JasperTan97, 2021-07-09:
@kailaili , I am quite curious at why you input such high acceleration noise and bias. I found that using the measured noise and bias of my imu (xsens mti) causes lili_om_rot to fail quickly but following your values, it runs very well.

Why do you think this is so?