KIT-ISAS/lili-om #40: loop closure performance

greymfm, 2021-07-12:
Hello,

I tried your code with some larger recording (Livox Horizon with internal IMU). It works except the loop closure doesn't work at the end of the recording.

[image]

[image]

I have uploaded the bag file here (recorded using livox_lidar_msg.launch):
https://drive.google.com/file/d/1NheqSfHnh1E00umY0no6I3AqigL7WDFK/view?usp=sharing

Maybe there is something I can try to tune the loop closures? :-)

Thanks,
Alexander

swjtuyang, 2021-09-28:
any update？
====
PS: I plan to add cm-precise RTK-GPS factors (already available as LiDAR time-synchronized NavSatFix messages in the above recording) to the global graph pose optimization to make 'lili-om' even more robust ;-)

kailaili, 2021-09-28:
Hi, you may change the parameters related to loop closure in the yaml file, e.g., raise the lc_search_radius. But yeah, adding additional external factors would be very good to achieve guaranteed robustness as sometimes the drift could be too large for closing the loop. I'm looking forward to seeing your further development. :)

schoeller, 2022-08-28:
> PS: I plan to add cm-precise [RTK-GPS factors](https://github.com/devbharat/gtsam/blob/master/gtsam/navigation/GPSFactor.h) (already available as LiDAR time-synchronized NavSatFix messages in the above recording) to the global graph pose optimization to make 'lili-om' even more robust ;-)

@greymfm 
Is there any news on this?