KIT-ISAS/lili-om #46: parameter tuning

bugdeveloper0, 2022-01-20:
hi, i am troubled in getting the desired result from your code and dataset. what parameters should i tune in run_fr_iosb_internal_imu.launch and config_fr_iosb_internal_imu.yaml when playing FR_IOSB_Short.bag. The end-to-end error in your paper is 0.25 m or lili-om, but i have the result about 2m. looking forward to your reply!

kailaili, 2022-01-20:
Hi, *_fr_iosb_internal_imu.yaml is for the internal IMU, whereas the one in the paper refers to an external IMU (MTi-670).  Apparently one could expect that the performance of the internal IMU is worse than the external one.

closed 2022-01-20 by kailaili

bugdeveloper0, 2022-01-20:
thank you for your reply. It fails when i play other bagfiles such as hku campus provided by hku mars lab. what should i do to make the program work normally for other recorded bagfile?

kailaili, 2022-01-20:
Usually the calibration should be renewed, and some further tuning of the parameters is essential.

bugdeveloper0, 2022-01-20:
is the program already compatible with avia?