KIT-ISAS/lili-om #50: Compatible with IntelRealsense L515?

luiscastro1995, 2022-04-01:
Hi,

I have been trying to adapt the package to work with Realsense L515, however I am getting the following warnings:
```bash
Failed to find match for field 'intensity'.
Failed to find match for field 'normal_x'.
Failed to find match for field 'normal_y'.
Failed to find match for field 'normal_z'.
Failed to find match for field 'curvature'
```
The pointcloud published by realsense has the following fields "x", "y", "z" and "rgb". Is there a way to obtain the 'intensity', 'normal_x', 'normal_y', 'normal_z' and 'curvature'. I have looked into "FormatConvert.cpp" where the correct pointcloud is built but I don't know if it is possible to do the same for the resulting pointcloud of L515. Are all those fields important?

Thanks!

hichamhendy, 2022-05-17:
I alsotry to do it and have more problem! am not sure what exactly should be adjusted in order to be integrate Realsense L515.

Have you reached any progress?

kailaili, 2022-07-15:
Hi, sorry for the late response. The current frontend does not support L515 as it as different scan pattern. You may try investigating its own scan pattern and extract edge and normal feature points as suggested in the paper.  The backend fusion block should be applicable afterward.

closed 2022-07-15 by kailaili